muka/go-bluetooth #141: is it useable for android?

MatejMagat305, 2021-09-11:
Hi I m coding in https://github.com/fyne-io/fyne, and it can run on android, it this library suitable for android?

muka, 2021-09-12:
Hi, nice project! 
This library uses the dbus APIo of bluez. I guess android uses bluez for kernel communication but not sure about it. If you find it works i would be happy to add references to your project. Thanks

MatejMagat305, 2021-09-12:
So I will try it, and I will reference

MatejMagat305, 2022-04-08:
I m sorry it look like android need jni ............

MatejMagat305, 2022-04-08:
Can I ask you? You have some experiences about bluetooth, I plan that I will create simple golang-jni interface to create server and client and sending data between both ........, do you planing some like that or not? Because I do not want duplicate work .......... and what is "minimal" interfaces to try embeded to your code ?

closed 2022-06-04 by muka

jizhihaoSAMA, 2022-07-13:
Well, due to uncertainty and ignorance of Android and Bluetooth, I am sorry that I selected Python-Kivy and Python-bleak to solve this.