muka/go-bluetooth #143: MapToStruct: Field not found: MTU

Arsen6331, 2021-10-24:
I have updated my computer (running Archlinux), and have begun receiving the following error: `MapToStruct: Field not found: MTU`

BlueZ version: `5.62-1`
Dbus version: `1.12.20-1`
go-bluetooth version (from go.mod): `v0.0.0-20210812063148-b6c83362e27d`

This issue seems to be caused by the BlueZ `5.62` update as downgrading the package makes the issue disappear.

tomaszkubacki, 2021-11-02:
I've got similiar issue  output is:

ERRO[0000] Error: MapToStruct: Field not found: Roles   
exit status 1

(default - current Ubuntu 21.10)
Bluez version : 5.60

OrgansWithoutBodies, 2021-11-21:
https://patchwork.kernel.org/project/bluetooth/patch/20210826164211.2936133-1-david@lechnology.com/ the API change in question that caused this afaict

muka, 2021-11-22:
Merged PR #145 Please reopen if it still an issue

sbinet, 2022-01-23:
the fix from #145 hasn't been incorporated into `master`. was it on purpose?
(I was a bit confused to get the same error even though I did `go get -u github.com/muka/go-bluetooth@latest`)

muka, 2022-01-23:
If it is on 5.62 master may not be aligned as it usually align with ubuntu releases (21.10 ships 5.60). 

Can you try with branch `bluez/5.62` or regenerate the sources using this command `BLUEZ_VERSION=5.62 make all`

Thanks

sbinet, 2022-01-23:
I could fix the error by explicitly specifying the commit from the `bluez/5.62` branch.
my question was more on whether it was on purpose that fix wasn't merged into `master`.
and it seems it was, as you follow Ubuntu releases.
thanks :)

muka, 2022-01-23:
The code is generated from bluez docs, it is possible that the modification has been overwritten by code generation

sbinet, 2022-01-23:
it seems to me they live on 2 different branches.

```
* 9857391 (origin/master, origin/HEAD, master) added usage section
* 48824ca override MediaPlayer1.Track to an actual type, fix #133
| * 1c7f879 (HEAD -> v5.62, origin/v5.62) generated v5.62
|/  
* 62e38ae (origin/v5.60) update docs
* 7bf4032 - fix parser on broken properties - remove experimental from file names
*   b99792b Merge pull request #145 from OrgansWithoutBodies/master
```

to reiterate: `1c7f879` is working fine on my ArchLinux box. (well, the `MTU` issue is not there)

muka, 2022-01-23:
Thank you, I will try to standardize the releases!

andig, 2022-02-17:
I‘ve just had this reported in https://github.com/evcc-io/evcc/issues/2627. I‘m not sure how to proceed: downgrade to working commit or upgrade master? Upgrade underlying Alpine image?

muka, 2022-02-18:
On master you should find the version compatible with 5.60. There are also branch with different bluez version

muka, 2022-02-19:
Could you try to  use `branch/5.62` ?

https://github.com/muka/go-bluetooth/blob/bluez/5.62/bluez/profile/gatt/gen_GattCharacteristic1.go#L117

muka, 2022-02-19:
I added the versioning reference on README and also generated a version for 5.63. I cannot test it right now, feel free to open PR with improvements

andig, 2022-02-19:
> Could you try to use branch/5.62 ?

How do I do that?

    ❯ go get -d github.com/muka/go-bluetooth@bluez/5.63
    go get: github.com/muka/go-bluetooth@bluez/5.63: invalid version: version "bluez/5.63" invalid: disallowed version string

muka, 2022-02-20:
Good point, I added git tags and should play better with `go get`

andig, 2022-02-20:
For reference: https://github.com/golang/go/issues/51281

andig, 2022-02-20:
I think the problem I've had is that I've upgraded master from 

	github.com/muka/go-bluetooth v0.0.0-20211122080231-b99792bbe62a

to 

 	github.com/muka/go-bluetooth v0.0.0-20211227073548-985739196620

The first commit had `MTU`, the latest nicht. Not sure how that happened as it seems that `MTU` is a feature of newer bluez releases?

muka, 2022-02-20:
Master refer to 5.60 which may not have MTU?

andig, 2022-02-20:
Right. What *is* confusing is that master *had* MTU in an older commit: https://github.com/muka/go-bluetooth/blob/b99792bbe62aa31189a614a3115dc9328eabb9f2/bluez/profile/gatt/gen_GattCharacteristic1.go#L117

muka, 2022-02-21:
It's my fault, I merged a contribution with that part but other that was broken. Recent commits fixed that. Dbus versions need to be addressed but I need to find a proper way. Do you have an idea?

andig, 2022-02-21:
That's good because the failure is now explainable. I have unfortunately no suggestion about dbus- I'm only consuming this module without deeper Linux experience.

muka, 2022-02-22:
Glad it is solved, my question is about a good way of managing go library versions in respect to the underlying blueZ version.

andig, 2022-02-22:
Two options I could imagine:
- separate go-bluetooth and go-bluetooth/bluez modules with different versions OR
- go-bluetooth gets a version policy of supported bluez versions (assuming you can support multiple ones at the same time). One release per bluez version with patches to previous versions per policy (i.e. only last 3 bluezs will be supported).

muka, 2022-02-22:
Thank you, I will evaluate the effort for those approaches. It's definitely
something to address. Please close if resolved.

sent from my phone

andig, 2022-02-22:
Resolved for me per https://github.com/muka/go-bluetooth/issues/143#issuecomment-1047245068

closed 2022-02-23 by muka