muka/go-bluetooth #146: Add send and recieve receive

MatejMagat305, 2021-11-25:
It is a little bit silly request, but it would be really helpful (at least for me) when you add something similar to server / client, but with sending and receive Massage, because I really do not know where I should give it.................., thank

muka, 2021-11-26:
Hi @MatejMagat305 try starting  from here https://github.com/muka/go-bluetooth/blob/master/_examples/cmd/service.go and see how it goes

MatejMagat305, 2021-11-26:
I understand that I could write according pictures, but I m not sure......, whether I can combinate.........
[image]
[image]
..

muka, 2021-11-26:
Here you are mixing two different things. You need a service exposing characteristics on the server side that can receive and send data. The client can then connect and send receive. Try looking into the Bluetooth specs for am overview. Probably you'll need also two Bluetooth cards one to serve one for client

MatejMagat305, 2021-11-26:
I m sorry for many question, but where https://github.com/muka/go-bluetooth/blob/master/examples/service/client.go are data sending?

muka, 2021-12-27:
Sorry, personally I do not have such example code! If you find a way please consider adding an example to the library with a PR as it would help other users also!

Closing, please reopen if you have updates. Thanks

closed 2021-12-27 by muka