muka/go-bluetooth #147: Properties.GetAll org.bluez.GattDescriptor1: No such interface 'org.bluez.GattDescriptor1'

Chillimeat, 2021-12-10:
I tried to get the description of Bluetooth, but prompted the following error? What should I do with it?
bluez version 5.55
ubuntu 18.04

IanSmith123, 2021-12-26:
remove file  
```
rm -rf bluez/profile/battery/gen_BatteryProvider1 [experimental].go
rm -rf bluez/profile/battery/gen_BatteryProviderManager1 [experimental].go
```
then you can follow the [example](https://github.com/muka/go-bluetooth#running-examples)
```
cd _examples
go run main.go
# print available example commands
# Example discovery
go run main.go discovery
```

muka, 2021-12-26:
The library is broken due to some changes in the bluez API itself. I am working on an upgrade, should be ready in a few days

muka, 2021-12-27:
Should be fixed on latest master. Closing, please reopen if you have updates. Thanks!

closed 2021-12-27 by muka