muka/go-bluetooth #150: Is bluetooth device emulation/advertisement usable?

nanoscopic, 2022-02-01:
The main readme says that "Expose bluetooth service from go code" is unstable. Is this still currently true? Could you elaborate on what sort of unstability has been noticed and what to expect?

This seems to be a core feature of tinygo bluetooth that uses this library, and if there are issues with it I would like to know what the issues are before deciding to use it.

muka, 2022-02-01:
If you decide to go for it and have feedback, PR and suggestions are always welcome!

nanoscopic, 2022-02-01:
Thanks. My intent is to use this by way of tinygo bluetooth with NRF52840 usb dongle to make virtual bluetooth keyboard/mouse for an iOS device farm.

The rest of the project is already in Go, so I wanted the portion of it for this to use Go as well. I'll be trying it out and if it works well will integrate its use into the larger project.

If I notice any way to improve it I'll be happy to contribute both feedback and/or PRs. Thanks for the open welcome.

closed 2022-02-01 by nanoscopic

muka, 2022-02-01:
Ok, good to know! I had a test setup but with different use-case, maybe is helpful https://github.com/muka/pi-wifi