muka/go-bluetooth #174: Missing tags for version 5.64 and 5.65

ChromaMaster, 2022-10-04:
Hi, are the versions that support bluez 5.64 and 5.65 committed and ready to use? If so, could you create the tags `bluez-5.64` or `v5.64` and `bluez-5.65` or `v5.65` so those versions can be pinned in the go modules? 

Thanks

muka, 2022-11-09:
Yes, could you provide a PR or a reference branch to merge with support for those releases?

ChromaMaster, 2022-11-10:
I guess:
- Branch: https://github.com/muka/go-bluetooth/tree/bluez/5.64 (commit 1d8857e3b268676077765e86c45bfc7a7d1758a9) would be `bluez-5.64` or `v5.64` 
- Branch: https://github.com/muka/go-bluetooth/tree/bluez/5.65 (same commit as the above 1d8857e3b268676077765e86c45bfc7a7d1758a9) would be `bluez-5.65` or `v5.65`

muka, 2022-12-13:
Done, sorry for the late update

closed 2022-12-13 by muka

ChromaMaster, 2022-12-14:
@muka Hi, it's possible that you didn't push the tags? I can't see them

muka, 2022-12-16:
Ohh yes, my fault. Just created the branches. TODO: automate new bluez releases