muka/go-bluetooth #188: Please generate version bluez-5.66.json

dezi, 2023-07-27:
Hello,

i get the following warning:

level=warning msg="MapToStruct: invalid field detected *device.Device1Properties.Bonded"

I guess the device property "Bonded" was introduced by bluez 5.66 and I like
to use this field. (Important for BLE devices)

I would be very happy, if this version is generated.

Regards,
dezi

andig, 2023-10-20:
Friendly ping @muka would that be possible?

andig, 2023-10-20:
I have no success running this under docker with bluez installed in order to create a PR:

```
go run ./srcgen/main.go full --debug
FATA[0000] chdir ../src/bluez/doc: no such file or directory
```

Unfortunately I don't have access to a Linux machine :/

andig, 2023-10-20:
Actually: there is a PR: https://github.com/muka/go-bluetooth/pull/186