muka/go-bluetooth #197: Test cases depend on host system bluetooth stack.

zelch, 2024-01-15:
Currently, I have a decent number of test failures, because my system isn't exactly what the test cases expect, at least some of them are due to that.

This poses a bit of a problem, given that ideally we would be able to run all tests successfully in a container without access to the host hardware, so we could run them during CI.

I am very much open to suggestions on how to manage this.

In a perfectly ideal world, it would be a docker setup with a real bluez stack, just with enough emulated hardware to pass all of our test cases.

I have a sneaking suspicion that this might not be a trivial undertaking.

zelch, 2024-01-15:
Or maybe it's possible, just horribly undocumented:

https://github.com/ukBaz/python-bluezero/issues/142

This at least makes it seem like a path forward might be technically possible.

I don't suppose anyone else is up to trying? :)

muka, 2024-01-17:
Yes you are right and would be a great addition. Honestly I never tried to have something like that working. 

A PR is welcome as always!  :1st_place_medal: